kevancress/MeasureIt_ARCH #262: AreaDimension BUG?

haukj, 2023-03-28:
AreaDimension wont show text value and - gives error when trying to export to SVG, DXF

-----------------------------------------------------------------------------------
11
ERROR: Odd Number of Coords, injecting padding to preserve other lines
"4.00 m²" on Cube has no textcard, failed to draw_text
-----------------------------------------------------------------------------------
[image]

kevancress, 2023-03-31:
Hey @haukj 

Thanks for reporting. Just pushed a change that should fix this 12604c11723b84ff1ce88b685c507fe16f7e1fef.

Give the latest development version a test and let me know if it works for you!

haukj, 2023-04-14:
Hi, I still get the same error. Both on my Mac (M1) and Windows PC. The area text is not drawn, there is just a leader pointing to scene coordinates 0,0,0...

kevancress, 2024-04-26:
@haukj @Huronsen Could one of you post an example file, I'm not able to reproduce this issue in Blender 4.0 with the latest development version of the addon.

Area's and area text require the scene to have an active camera, removing the active camera is the only way I've found to make the area dimension text not show up

Huronsen, 2024-04-26:
Hi Kevan

Yes, your right. It doesn't work only when there is no camera!
In an older version, there was aan option to align it to the view. Since I couldn't find that option i thought it is reffering to this former one. If that option is gone in the current version, it would be great to have it back :)