kevancress/MeasureIt_ARCH #263: can't active plugin on 3.5 on m1 Mac

PeterEller, 2023-04-02:
Using the latest version of measureIt_ach works fine on blender 3.4 but 3.4 crashes too often on my arm64 mac
3.5 is much more stable but when I try to install the plugin on 3.5 I just get errors

"Traceback (most recent call last):
File "/Applications/Blender.app/Contents/Resources/3.5/scripts/modules/addon_utils.py", line 333, in enable
mod = import (module name)
File "/Users/petereller/Library/Application Support/Blender/3.5/scripts/addons/Measurelt_ARCH-development/ init from . import measureit_arch_main
File "/Users/petereller/Library/Application Support/Blender/3.5/scripts/addons/Measurelt ARCH-development/measur from measureit_arch_geometry import clear_ batches, update_text, draw3d loop, preview_dual, check obj_vis File "/Users/petereller/Library/Application Support/Blender/3.5/scripts/addons/Measurelt_ARCH-development/measur
triShader = gpu.types.GPUShader(
Exception: Shader Compile Error, see console for more details"

PeterEller, 2023-04-26:
Thank You! That worked!!
I have been exporting all my models to FBX and doing all my dimensions is rhino 
I was about to give up 

cheers!!

buildxyz-git, 2023-11-03:
Thank you, this was very helpful! 

To be clear, are you leaving the backend setting in OpenGL? If I select Metal after activating the add-on, the MeasureIt-ARCH toolbar disappears on the next load of Blender. Is there just no support for Metal and MeasureIt-ARCH?

PeterEller, 2023-12-12:
this doesn't work anymore in blender 4

kevancress, 2023-12-17:
Hey all,

sorry for the slow response on this. I've just now finished moving all of the drawing code in the addon from Blender .bgl API, which was OpenGL specific, over to the new .gpu API which should support Metal as a back end. I don't have a MacOS device to test this on though. If someone could try the latest development version of the addon, and let me know if it enables on an M1 Mac with the Metal backend that would be much appreciated!

kevancress, 2023-12-17:
There may still be some shader tweaks needed to get things working due to shader compilation differences between Metal and OpenGL...

PeterEller, 2023-12-17:
Sorry it still doesn’t work
Here is the error I get

￼
Peter Eller  
MODELLER | MODERN FURNITURE
Direct: +1 (604) 736-5797 | ***@***.***
1157 E 23rd Vancouver British Columbia
www.modeller.ca <https://www.modeller.ca/>

haukj, 2023-12-17:
I’m more than willing to help you out :)

Kjetil Haughom

BIM Koordinator

 [Outlook-E-post sig.png]

Mynd Entreprenør AS

Orgnr. NO995 962 373MVA

Dokkvegen 11, 3920 Porsgrunn

Direkte: (+47) 950 80 650  Sentralbord: (+47) 479 22 300
Web: www.mynd.no<http://www.mynd.no/>

kevancress, 2023-12-18:
@PeterEller @haukj 

Hmm okay, looks like I'll need to do some shader compilation tests on an m1 mac. Does switching to the OpenGL backend work in 4.0 with the latest development build of the addon? Hoping that still works as a backup while I track down what in particular in the shaders is causing an issue with Metal

PeterEller, 2023-12-18:
I don't think OpenGL is an option anymore on Blender 4 with Apple silicon ?

You should ask Santa to bring you one of those new M3 MacBooks ;-) The performance in blender is stunning!

I tested out the Measurable add-on it doesn’t work in v4 either. So, I've been using Rhino for dimensions recently.  UGH

I appreciate your good work. you should set up a buymeacoffee.com

I just used It recently to send some money to the developer who makes a plugin I use for Obsidian "Database folder"

https://github.com/RafaelGB/obsidian-db-folder￼
RafaelGB/obsidian-db-folder: Obsidian Plugin to Allow Notion like database based on folders
github.com

￼
Peter Eller  
MODELLER | MODERN FURNITURE
Direct: +1 (604) 736-5797 | ***@***.***
1157 E 23rd Vancouver British Columbia
www.modeller.ca <https://www.modeller.ca/>

haukj, 2023-12-19:
Mac OS M1: There is no option to switch from Metal to OpenGL in Blender 4.0 anymore

haukj, 2023-12-20:
Latest error when trying to activate in Blender 4.1 : Traceback (most recent call last):
  File "/Applications/Blender 41.app/Contents/Resources/4.1/scripts/modules/addon_utils.py", line 364, in enable
    mod = importlib.import_module(module_name)
  File "/Applications/Blender 41.app/Contents/Resources/4.1/python/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Users/username/Library/Application Support/Blender/4.1/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 78, in <module>
    from . import measureit_arch_main
  File "/Users/username/Library/Application Support/Blender/4.1/scripts/addons/MeasureIt_ARCH-development/measureit_arch_main.py", line 33, in <module>
    from .measureit_arch_geometry import clear_batches, update_text, draw3d_loop, preview_dual, check_obj_vis
  File "/Users/username/Library/Application Support/Blender/4.1/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 101, in <module>
    triShader = gpu.types.GPUShader(
Exception: Shader Compile Error, see console for more details

kevancress, 2023-12-21:
@PeterEller @haukj Just pushed a commit that removes an unused variable in the shader that seems to be causing the error. Not sure if this is what's tripping up the compiler, haven't had a chance to test on a mac yet (might have access to one between Christmas and newyears)

If you can give it a try and let me know if that fixes things or changes the error at all that would be great!

Thanks!

Ps. @PeterEller  I did set up a kofi page for donations a while back (https://ko-fi.com/measureit_arch) but always feel a little guilty sharing it since my development time is so fragmented these days

haukj, 2023-12-21:
Sure @kevancress :) Just downloaded your latest commit.
Tested on both B3.6 and 4.1

Blender 3.6 With Metal Backend enabled:
Error message:
2023-12-21 21:21:06.265 Blender[65537:4319735] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.
Info: Updated sidebar tabs
M_ARCH import modules
Traceback (most recent call last):
  File "/Applications/Blender.app/Contents/Resources/3.6/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/Users/kjetilhaughom/Library/Application Support/Blender/3.6/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 78, in <module>
    from . import measureit_arch_main
  File "/Users/kjetilhaughom/Library/Application Support/Blender/3.6/scripts/addons/MeasureIt_ARCH-development/measureit_arch_main.py", line 33, in <module>
    from .measureit_arch_geometry import clear_batches, update_text, draw3d_loop, preview_dual, check_obj_vis
  File "/Users/kjetilhaughom/Library/Application Support/Blender/3.6/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 101, in <module>
    triShader = gpu.types.GPUShader(
Exception: Shader Compile Error, see console for more details

Blender 4.1
Error message:
Traceback (most recent call last):
  File "/Applications/Blender4.app/Contents/Resources/4.0/scripts/modules/addon_utils.py", line 364, in enable
    mod = importlib.import_module(module_name)
  File "/Applications/Blender4.app/Contents/Resources/4.0/python/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 78, in <module>
    from . import measureit_arch_main
  File "/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/measureit_arch_main.py", line 33, in <module>
    from .measureit_arch_geometry import clear_batches, update_text, draw3d_loop, preview_dual, check_obj_vis
  File "/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 101, in <module>
    triShader = gpu.types.GPUShader(
Exception: Shader Compile Error, see console for more details

haukj, 2023-12-21:
@kevancress  Is it still trying to access the old glsl files?

kevancress, 2023-12-21:
@haukj Ahh okay that stray variable wasn't the issue then... 

I tried another minor change, could you give that a try as well? 

If you're still getting "triShader = gpu.types.GPUShader(Exception: Shader Compile Error, see console for more details" as the last two lines of the error then would you mind trying to run blender from the terminal with the arguments  "--debug-gpu --gpu-backend metal" trying to enable the addon and then sending the full terminal result to me?

that should give some more specific information about what exactly is causing the shader compilation to fail.

The addon does still have its custom shaders written in .glsl, but Blender's GPU module has a system that cross compiles the shaders to work with the various graphics backends (OpenGL, Vulkan, Metal), so in theory .glsl should be fine (most of blender's internal shaders are still .glsl too I believe), but there must be somthing in the .glsl that the compiler can't deal with when converting to metal

haukj, 2023-12-25:
Sorry for the delayed response. Here's  the error message when running Blender with gpu debug flags (latest version of measureIT)

M_ARCH import modules
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader_generator.mm:885 generate_msl_from_glsl: Unable to compile shader 0x17974d408 'pyGPUShader' as no create-info was provided!
Traceback (most recent call last):
  File "/Applications/Blender4.app/Contents/Resources/4.0/scripts/modules/addon_utils.py", line 364, in enable
    mod = importlib.import_module(module_name)
  File "/Applications/Blender4.app/Contents/Resources/4.0/python/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 78, in <module>
    from . import measureit_arch_main
  File "/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/measureit_arch_main.py", line 33, in <module>
    from .measureit_arch_geometry import clear_batches, update_text, draw3d_loop, preview_dual, check_obj_vis
  File "/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/measureit_arch_geometry.py", line 101, in <module>
    triShader = gpu.types.GPUShader(
Exception: Shader Compile Error, see console for more details

WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 2)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 2)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 2)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
[... cut ...]
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'

kevancress, 2023-12-26:
Ahh okay that’s actually really helpful!! I think that means that providing some more explicit shader creation info through the gpu api should resolve this. I’ll try that out today or tomorrow, thanks!!

kevancress, 2023-12-28:
Hey all,

Unraveling this bit by bit going through the documentation for [Cross Compilation](https://wiki.blender.org/wiki/Source/EEVEE_%26_Viewport/GPU_Module/GLSL_Cross_Compilation) and the [ShaderCreateInfo](https://docs.blender.org/api/current/gpu.types.html#gpu.types.GPUShaderCreateInfo). I think I've got a handle on how to set up the basic shaders, but line drawing uses a custom geometry shader... which doesn't seem to be directly supported by Metal...

I'll have to do some thinking on how I can re-write the drawing systems without a geometry shader stage, or see if I can find a way to make it work with metal which will take a bit of time. I'll keep at it this week and let you know how it goes

kevancress, 2023-12-28:
Hey all,

I've made some progress on the re-write. Still a lot missing as far as line drawing is concerned (lineweights, dashes etc aren't re-implemented yet), but I wanted to ask if you could check if the new shaders are compiling on metal?

@haukj would you mind downloading the latest development version of MeasureIt_ARCH and trying to enable it with (with Blenders gpu debug flags enabled)

I've re-written all the custom shaders and initialized them with the proper create info declared for cross compilation, so this should enable with Metal now... if it does enable successfully then I'll keep working on getting the line drawing functionality back up to par with where it was previously.

Thanks!

haukj, 2023-12-30:
@kevancress Sure! I tried to enable the latest version, but it resulted in another error: Traceback (most recent call last):
  File "/Applications/Blender4.app/Contents/Resources/4.0/scripts/modules/addon_utils.py", line 401, in enable
    mod.register()
  File "/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 245, in register
    bpy.utils.register_class(cls)
RuntimeError: Error: Registering panel class: parent 'SCENE_PT_unit' for 'SCENE_PT_MARCH_Units' not found

I opened blender without having GPU debug flags enabled...I don't think you need these to correct this error

kevancress, 2023-12-31:
huh... Well that seems like the shaders are compiling, but odd that were getting a panel class registration error on Mac and not Windows, especially since the parent class that cant be found ('SCENE_PT_unit')is one of Blender's internal UI classes...

Any other errors appearing in the console? Sometimes panel registering errors happen downstream of other issues.

kevancress, 2024-01-03:
Hey @PeterEller @haukj

Got my hands on an M1 Mac Mini to do some debugging tonight. Found there was still a shader error with the line overlay colors. Think I've got that fixed now.

At least on my end I've got the addon successfully enabling with on Blender 4.0.2 !

Let me know if its working for you!

Still got lots to do to get the Line Groups drawing propperly with these new shaders, but I should be able to make faster progress on that now that I have access to a Mac to test on.

Cheers!

kevancress, 2024-01-03:
Alright, most of the line drawing features are restored now in the latest commit:

[image]

Line Thickness, Dashes, and Colors are all working again! I still need to recalibrate the line and point thicknesses so that they match, fix the corners on thick lines, and make sure the new line weight system is consistent between the raster and vector outputs, but we're nearly there.

Let me know if this is all working on your end!

kevancress, 2024-01-03:
Fixed another minor issue with thick lines on rotated objects 4a74ba056b53c74115a0bce7de3a5cfcc9d2ec00

haukj, 2024-01-03:
@kevancress I got the same error again when trying your latest version. And I started to suspect that maybe some of the other enabled add-ons where messing things up because of all the big changes, so I started to disable them, and what do you know! - When I disabled Rigify (the version that comes built-in), the add-on could be enabled, but there was old junk in the UI from the rigidify add-on. I restarted Blender aaaand now it won't be enabled again...strange....Maybe there is some classes or calls that coincidentally are named the same? (FYI  - I have BlenderBIM enabled...) Debug Log: Modules Installed (MeasureIt_ARCH-development) from '/Users/kjetilhaughom/Downloads/MeasureIt_ARCH-development.zip' into '/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
M_ARCH import modules
Push constants have a minimum supported size of 128 bytes, however the constants added so far already reach 144 bytes. Consider using UBO.
Push constants have a minimum supported size of 128 bytes, however the constants added so far already reach 208 bytes. Consider using UBO.
Push constants have a minimum supported size of 128 bytes, however the constants added so far already reach 224 bytes. Consider using UBO.
Push constants have a minimum supported size of 128 bytes, however the constants added so far already reach 248 bytes. Consider using UBO.
Push constants have a minimum supported size of 128 bytes, however the constants added so far already reach 264 bytes. Consider using UBO.
Push constants have a minimum supported size of 128 bytes, however the constants added so far already reach 288 bytes. Consider using UBO.
Push constants have a minimum supported size of 128 bytes, however the constants added so far already reach 304 bytes. Consider using UBO.
Push constants have a minimum supported size of 128 bytes, however the constants added so far already reach 312 bytes. Consider using UBO.
M_ARCH register
Exception in module register(): /Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/__init__.py
Traceback (most recent call last):
  File "/Applications/Blender4.app/Contents/Resources/4.0/scripts/modules/addon_utils.py", line 401, in enable
    mod.register()
  File "/Users/kjetilhaughom/Library/Application Support/Blender/4.0/scripts/addons/MeasureIt_ARCH-development/__init__.py", line 245, in register
    bpy.utils.register_class(cls)
RuntimeError: Error: Registering panel class: parent 'SCENE_PT_unit' for 'SCENE_PT_MARCH_Units' not found

WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 2)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 2)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 2)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_shader.mm:624 uniform_int: Texture uniform location re-mapping unsupported in Metal. (Possibly also bad uniform location 3)
[... cut ...] ay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'
WARN (gpu.debug.metal): source/blender/gpu/metal/mtl_batch.mm:164 prepare_vertex_binding: MTLBatch: Could not find attribute with name 'vclass' (defined in active vertex format) in the shader interface for shader 'overlay_extra_groundline'

kevancress, 2024-01-03:
Okay... Well at least we know its not a shader issue this time! 

I just took a look at the code for the latest version of BlenderBIM, looks like it makes some pretty extreme changes to the Scene Properties UI which involves unregistering a lot of blenders default UI classes and replacing them with custom ones... fairly certain this is the cause of the issue since the 'SCENE_PT_unit' class gets unregistered when initializing BlenderBIM and replaced with one called 'SCENE_PT_unit_override'.

I can try and code up a check for this, and add a fall back that adds MeasureIt_ARCH's unit properties to its own panel, rather than as a subpanel of the typical unit settings, but other than that there's not much I can do if the parent class is getting removed by another addon...

either way since this is a separate issue I'm going to open a new thread for it.

kevancress, 2024-01-17:
Hey @PeterEller @haukj

Since this seems to be pretty stable on the M1 mac I've been testing on I'm going to close this issue for now. Feel free to re-open or start a new issue if things still aren't working correctly

closed 2024-01-17 by kevancress

PeterEller, 2024-01-17:
Yes, its working well for me now 
Thanks!

Peter Eller  
MODELLER | MODERN FURNITURE
Direct: +1 (604) 736-5797 | ***@***.***
1157 E 23rd Vancouver British Columbia
www.modeller.ca <https://www.modeller.ca/>