kevancress/MeasureIt_ARCH #264: Display issues on Blender 4.0.0

robouden, 2023-05-30:
I know blender V4.0.0is alpha. But the dimensions do not display on my setup. Any solutions?

- Blender 4.0.0,
- MSI GS-66
- Linux Mint

Regards,
Rob Oudendijk

kevancress, 2023-05-31:
Hey @robouden.

Thanks for reporting. Blender 4.0 has depreciated the .bgl API that MeasureIt_ARCH uses for nearly all of its viewport and raster drawing.

Updating to work with the .gpu API in Blender 4.0 will take a pretty extensive re-write of all of the add-on drawing code. So it may be some time before the add-on works with Blender 4.0. Best to stick to 3.5 stable for now if you want to use MeasureIt_ARCH

realstevewarner, 2023-12-13:
Kevan, thanks for the info on Blender 4.0.  MeasureIt Arch has become critical to my day-to-day work (architectural planning). I can't update to Blender 4.0 if MeasureIt Arch isn't working.  Have you considered taking the addon commercial?  There's really nothing on the market that offers comparable features.

kevancress, 2023-12-17:
Hey @robouden @realstevewarner 

Sorry about the delay on this! I've just finished the updates to get everything moved over to the .gpu API so the addon should be working with Blender 4.0 now!

There's still a couple render issues with raster linework in perspective views... I'll try and get those resolved over the next few weeks!

@realstevewarner, no plans to take the addon commercial, my goal with this has always been to keep these tools accessible to as many people as possible!

realstevewarner, 2023-12-17:
Thanks, Kevan! I'm off to test it now! Fingers crossed!

realstevewarner, 2023-12-18:
Hey Kevin, it works great! So happy to have this working in 4.0.  Only thing I noticed is that auto alignment isn't working. Do you know if that's possible with the 4.x changes?  Thanks again for the hard work on this!

kevancress, 2023-12-18:
@realstevewarner Great! Glad that this is working in 4.0 now! Which Auto alignment isn't working? Annotation text alignment and Dimension alignment seems to be behaving the same for me. Could you open a new issue for this. I'm going to close this one now that the 4.0 update is complete.

Thanks!!

closed 2023-12-18 by kevancress