kevancress/MeasureIt_ARCH #265: I cant find hide units Or I wish Units as NONE would have precision

Blendersbrontobyte, 2023-08-14:
Ether way I hope "hide units" in in settings settings  does not disappear.  And I hope to see units set as NONE have precision.  Ethier one of those would be great.   
 Thank you

kevancress, 2023-12-17:
Hey @Blendersbrontobyte

Hide units is still available under the scene settings 
[image]

No plans to remove this

closed 2023-12-17 by kevancress

Blendersbrontobyte, 2023-12-17:
Keven,
Thank you – your program opens doors for a lot of people. thank you for your reply.
Don Matteson
Cody Wyoming

image.png (view on web)<https://github.com/kevancress/MeasureIt_ARCH/assets/16860574/bdcb7859-5bc2-4a3c-ab95-96b48e8a0e55>