kevancress/MeasureIt_ARCH #266: Render Preview error in Blender 3.6

dimitarsp, 2023-08-24:
Hi, love the work you're doing! I've noticed an error that in 3.6, in render preview mode with Cycles and Measure it arch enabled, the rendering seems to be stuck at re-rendering only 1 sample and can't get past it.

kevancress, 2023-12-17:
Hey @dimitarsp 

Sorry for the slow response on this, had a very busy last couple of months, getting caught up on maintenance here over the Holiday break!

Unfortunately I think this is a limitation of how Cycles interacts with the python draw handler API. MeasureIt_ARCH draws every time Blenders Viewport updates, which seems to be ~30-60 times a second, unfortunately Cycles seems to interpret MeasureIt_ARCH's draw call as an update to the scene, which causes it to restart the preview rendering...

Not sure if this is solvable on the Python side, maybe there's another way to handle python viewport drawing that I'm not aware of that works better with cycles?

ktjo, 2023-12-23:
First of all, thank you for all your work on this very useful addon!

I think I've managed to track down what triggers the continuous redraw, to line 3983 in measureit_arch_geometry.py.
[https://github.com/kevancress/MeasureIt_ARCH/blob/321570b0e6d993a48e2b17703008491cc8b6524d/measureit_arch_geometry.py#L3983](url)

Adding a condition to this assignment seems to avoid the update loop.
```python
if dim.fontSize != dimProps.fontSize:
    dim.fontSize` = dimProps.fontSize
```

Another issue might be on line 513
[https://github.com/kevancress/MeasureIt_ARCH/blob/321570b0e6d993a48e2b17703008491cc8b6524d/measureit_arch_geometry.py#L513](url)
In my testing the `text_field_invalid` condition is always true, perhaps there is missing some set/reset of the `is_invalid` flag in the code?

kevancress, 2024-01-08:
@ktjo thanks for catching this! I'd always assumed that having a python 'Post_View' draw handler enabled would cause a cycles re-draw regardless, but just did some tests with a basic post view draw and your right, it seems like it is possible to not interrupt the cycles render samples!

I'll take a look at my assignments and flags to see if I can get this working with the cycles render preview!

kevancress, 2024-02-28:
@dimitarsp @ktjo 

Sorry this took so long, I think I've got this working well with cycles now as of this commit 17fce9860a2d9622a5cd6114d1a30a04c93a0fbf. It seems to not interrupt the viewport preview render anymore!

Cheers
feel free to re-open if there's any other issues!

closed 2024-02-28 by kevancress