kevancress/MeasureIt_ARCH #268: Auto Aligned Text doesn't align in v0.5 with Blender 4.0

realstevewarner, 2023-12-18:
Hey Kevan!  In Blender 3.6 using MIA 0.4.6 using the Add Dimension - Aligned on an edge will create a measurement that will rotate and adjust to the 3D viewport's viewing angle.  In Blender 4.0 using MIA 0.5.0, the same process will add the measurement, but it will not auto-adjust to the angle of the 3D viewport.  

The attached video shows this in more detail with side-by-side comparisons.

https://github.com/kevancress/MeasureIt_ARCH/assets/104595011/454aae97-48ce-484f-be5d-75aa8f8ca221

kevancress, 2023-12-18:
Ahh okay! Yep this was actually an adjustment I made between MeasureIt_ARCH v0.4.6 and v0.5.0. I was finding the automatic alignment to the viewport challenging to work with when laying out larger plans and sections, and so I set everything to stay aligned to the View's Camera, rather than the 3D viewport.

I should be able to restore the old viewport alignment behavior if its desirable. Maybe I'll add a toggle in the options menu so that  viewport aligned or camera aligned can be set depending on the desired behavior for a scene.

kevancress, 2023-12-18:
Also thanks for the side by side. This reminds me I also need to make some adjustments to the default dimension style values to make the default a bit more visually coherent like the older version. I made some changes to how lineweights and arrow sizes get computed to have better consistency between raster and vector outputs, but never did a visual/aesthetic update to the defaults.

realstevewarner, 2023-12-19:
Ah! That makes sense about the larger layouts being a challenge with the auto alignment.  But yes, it would be extremely helpful (to me at least) to have that as an option which can be turned on or off.  

A lot of what I do at work is high level space planning.  Take a CAD plan or Google Earth view and rough out dimensions to determine if we can properly develop the area.  So a lot of it is just quick screen grabs from the 3D viewport and having the dimensions automatically align to my view makes this process extremely fast.  

[image]

I really appreciate the work you've done on this addon.  I know it's a labor of love, but I can honestly say it's my most frequently used Blender addon and your work on it has been a game changer for me.

kevancress, 2023-12-19:
@realstevewarner 

Okay this was a pretty easy fix, all the view alignment code was still in there, I was just bypassing it if a scene camera was defined.

Added a new option in the scene settings to default to viewport based alignment if desired.
[image]

Glad the addons been helpful for you!!!

realstevewarner, 2023-12-19:
Fantastic!  Just tested it and it's flawless!  Thanks for the quick fix on this!

kevancress, 2023-12-20:
Great!! Let me know if you find any other issues!