kevancress/MeasureIt_ARCH #270: Area Measurements don't display properly

realstevewarner, 2024-01-25:
There are two issues with Area Measurements.  The measurement numbers typically exist at the 0 location on the Z axis which causes them to appear "tinted" by the area color.  The normal way to resolve this is to offset the number in the Z axis slightly.  (This should be the default for an Area measurement).  However in the current build (pulled on 1/23/24) offsetting the measurement does not result in the numbers becoming black.  One or two might be black, but this is random and intermittent.  In previous builds, this was not the case, but it now means I have to render Areas with full color opacity as one pass, then adjust all areas and set color opacity to 0 so I get just the numbers.  

[image]

The second issue is with additional text alignment.  The text for area measurements is always Left Justified.  It would be nice to have the text alignment fixed.  Not as critical as the text offset issue, but still a nice to have.

Thank you!

kevancress, 2024-01-30:
@realstevewarner

hmm okay. I had to do an overhaul on the drawing API to work with Blender 4.0 and I must be doing something wrong with my alpha blending that's causing the semi transparent areas to not blend properly and tint the text.

Pushed a quick fix (c91d09482ce2302064ba80283a329deec013e3b3) that just forces text to draw after the areas, which should work in most cases, not super robust though, so I'll see if I can get a proper fix done later.

For Alignment. Yeah I need to do a big overhaul of how text alignment gets set, its always fiddly right now. I'll try to fix the dropdown for it in the area dimension settings so that it actually works!

realstevewarner, 2024-01-30:
Thanks, Kevan!  It's working great now.  I appreciate you doing a quick brute force fix!

[image]

As a side note, I used MeasureIt ARCH on my most recent project and finally got into the rendering side of things.  It's amazingly powerful.  When I tried rendering about a year ago, it gave me errors so I resorted to quick screen grabs and have been doing that ever since.  But I set up proper cameras and was amazed at how well rendering both to image and to SVG works.  I'm now getting true CAD level rendering out of Blender.  Thank you!