kevancress/MeasureIt_ARCH #271: Can't activate addon in Blender 4.0.1

FelixHob, 2024-02-19:
[image]

kevancress, 2024-02-22:
Hey @FelixHob,

Which version of the addon are you using? If you download the zip directly from the main github page it should work fine with blender 4.0+.

From the error message this looks like you're trying to install the v0.4 release, which wont work with current versions of blender.

kevancress, 2024-02-28:
@FelixHob 
Closing this for no since I can't reproduce the issue. If your still not able to activate the latest version of the addon with Blender 4.0 feel free to open another issue.

closed 2024-02-28 by kevancress