kevancress/MeasureIt_ARCH #272: Updating styles on updating from 3.6 to 4.02

skonato, 2024-03-02:
Hi Kevan,

Thanks a lot for the great addon!
I have updated Blender from 3.6 to 4.02 and the addon to 0.5.0.
All dimensions in the projects save under 3.6 were disappeared. I can see only their styles' names in the dropdown lists. However, I do not see these styles in the project settings and that is why I am not able to delete them.
Is it possible to delete old styles from dropdown lists?
If I create a style with the same name, I see this style in dropdown lists twice.

[image]

Thanks!
Konstantin

kevancress, 2024-03-05:
Hey @skonato 

If you use the "Delete All Styles" from the dropdown list next to the styles that should clear out any old styles, even if they're not visible. They really should still show up from version to version though, so I'll have to look into fixing that.

What version of the addon we're you using before you updated?

skonato, 2024-03-05:
Hi @kevancress 

Thanks a lot for your reply! I reinstalled Blender on another PC and simulated this error for you:
Test scene on the PC with 3.6.2/0.4.7:
[image]

The same scene opened on the PC with 3.6.2/0.4.7:
- no dimensions;
- no GrrenStyle in the global list;
- not possible to delete all styles;
[image]

The error message on attempt to delete all styles on the PC with 4.0.2/0.5.0:
[image]

Here is the Blender file example:
[TestExample.zip](https://github.com/kevancress/MeasureIt_ARCH/files/14499617/TestExample.zip)

kevancress, 2024-03-05:
@skonato That perfect, thanks for setting that up!

I'll take a look at this tonight and see if I can get this resolved!

skonato, 2024-04-23:
Hi @kevancress ,

Is there any hope that the bug will be fixed?

kevancress, 2024-04-23:
Hey @skonato 

Thanks for the nudge on this, got partway to a solution but hadn't finished it. Took another crack at it now, and I think I've got it working (seems to load the correct styles on the test file at least.)

Give the latest development version a check and see if its working for you!

skonato, 2024-04-26:
Hi @kevancress,

Thanks for your reply. I tried the same file once again with the latest downloaded version today and here are the results: 

no dimensions [NOT FIXED]
no GreenStyle in the global list [FIXED]
not possible to delete all styles[NOT FIXED]

[image]

kevancress, 2024-04-29:
@skonato 

Thanks for the update, I'll take another look

skonato, 2024-05-06:
Hi @kevancress,

Sorry, but I have the same issue in the latest Blender 4.1.1

kevancress, 2024-05-06:
@skonato

Sorry I haven't had a chance to get back to this. Did the update to 4.1.1 break the styles showing in the styles list again, or is it just the two remaining issues (dimensions not showing and style deleting) that are still broken?