kevancress/MeasureIt_ARCH #273: Getting started guide

neopostmodern, 2024-03-05:
Just stumbled upon this project and I *think* it'll be really useful for me, but I have a hard time getting started, i.e. rendering anything visible. Therefore I'd like to suggest to add a minimal walkthrough to the docs site.

Things I discovered I needed which I found unintuitive (not bad design, just in need of a hint for new users):
- You need to add lines manually (e.g. "Dynamic Line Group")
- Rendering happens via specific buttons, not Blender rendering
- Most render settings are in the scene tab, not the render tab

Things I haven't discovered yet:
- Setting a white background to actually see something
- Setting a customized drawing style throughout

TL;DR
It would be great to have a walkthrough how to create this picture from a blank project:
[image]

skonato, 2024-03-05:
It would be great to add something about hatches and section cuts to this list.

kevancress, 2024-03-05:
@neopostmodern @skonato 

Totally agreed! I've had an empty tutorial section in the documentation for way to long. I'll try and make some time to tackle this soon!

neopostmodern, 2024-03-06:
Great! Let me know when there is something to proofread/test and I'd be happy to give authentic feedback from a first-time user :)