kevancress/MeasureIt_ARCH #274: Size location

loonyproof, 2024-03-05:
Hi 
Can I somehow change the position of the text? Move size to red square. 
[image]

kevancress, 2024-03-05:
Hey @loonyproof 

Yeah I've been meaning to add better text placement tools for dimensions for a while, not easily possible at the moment though.

loonyproof, 2024-03-06:
In the previous version you could put a negative number (in front size) and then the text would move
Is it possible to do this somehow?
[image]

kevancress, 2024-03-06:
oh weird, I had no idea that would have worked that way with negative font sizes...

For now I've just setup a toggle for Dimension text positioning. Give the latest version a download and see if it works for you!

[image]

Middle positioning should break the dimension line to either side of the text box, but haven't set that up yet, top and bottom work though.

I'd really like to have a free placement option for text with a leader line pointing back to the dimension also, but that will take some more time.

Hope this works okay for now!

loonyproof, 2024-03-07:
thank you))
But for some reason it doesn't work for me(
If I click on this switch, and then click on the field, then it automatically switches again to the upper position.

https://github.com/kevancress/MeasureIt_ARCH/assets/110474910/b287f12d-80ea-448f-9bd6-90b1beb670fb

tessafk, 2024-10-24:
Hello! Would it be possible to add an option that would allow to rotate the dimension? For example in the case illustrated above, rotate the text 90 degrees to align it to the x axis. Thank you so much!