kevancress/MeasureIt_ARCH #275: Preview problem in Blender 4.02

ajo-maker, 2024-03-25:
Hi Kevan,
What a great add-on! Thank you for this.
I updated blender to version 4.02. And now I have problems with the line and dimension preview in the 3D viewport. Do you have any idea how I can fix this?
[image]
[image]
AJ

kevancress, 2024-04-23:
@ajo-maker 

The line weight differences were an intentional change to bring the raster and vector line weights into alignment. Not sure why the dashed lines aren't showing up. is the the latest development version of the addon, what system and graphics card are you using?

ajo-maker, 2024-04-28:
AMD Ryzen 9 5900HX with Radeon Graphics           3.30 GHz
nvidia geforce rtx 3070 laptop gpu

The dashed lines are indeed not displayed.
But the different line thicknesses are also not visible in Blender. In fact, everything is displayed on one line thickness. The specified thickness will only be visible in the generated SVG file.

kevancress, 2024-04-28:
Ahh okay, yep seeing that more clearly now in your image. Can you post the demo file that was generate that preview image? I'm not able to reproduce this on my end, and we're working with the same graphics card, so I'm not quite sure what could be causing this.

[image]

Can you also confirm the addon version info at the bottom of the settings in the scene panel. the latest version should look like this:

[image]

ajo-maker, 2024-04-28:
The version was the problem. I installed the latest version and it works perfectly now!
Thanks!
[image]