kevancress/MeasureIt_ARCH #276: Deleted angle still shows up

alexmro, 2024-03-26:
I deleted all dimensions and angles and one angle still appears. I delete the vertices and when I add them back, there it is again. I just can't get rid of that damn angle. Had to disable the addon

kevancress, 2024-04-23:
Hey @alexmro, could you post the file that has the broke angle dimension so I can try to investigate this, I haven't been able to reproduce this issue on my end.

Also if you could post the system (windows, macOS, linux) and graphics card your using that would be helpful.

No sure what could cause you to be unable to enter edit mode, nothing in this addon should impact switching modes.

kevancress, 2024-08-20:
Closing. Haven't been able to reproduce this issue. Re-open with the information requested above (System, addon version, blender version, demo file) if your still experiencing this issue.

closed 2024-08-20 by kevancress