kevancress/MeasureIt_ARCH #277: Display of slope and world z value

Huronsen, 2024-04-22:
Hi Kevan

I'm a landscape architect and I'm switching to Blender for my work.
This Add-On offers so many features I had in my mind for a very long time, like the option to display an edge. This one is so huge for my work and I was wondering since years that no software offers this. 
Since a couple weeks I was thinking about hiring a developer for creating a Add-On which basically does what this Add-On does. That was before I knew this Add-On. 
We as landscape architects go forgotten when it comes to software development, which mainly focuses on the needs of architects. So thats why im pushing Blender in my area of work.

I have two feature requests. 
1. An aligned dimension, like the one already existing. But it should display the slope in percentage with an arrow next to the pointing downwards the slope. This one would be huge, because during modelling, I could see at any time, if my transformation is within the tolerance of whats buildable. 
[image]

2. An annotation (best without a guide line) that displays the world z-value. This way, one could add the annotations only once, and would not have to rewrite them all the time. For my work, this one would be huge as well.
[image]

I'd really like to contribute to this, but my coding-skills are limited and this plugin is already way beyond my skills of writing a simple script. 
But I'd actually like to contribute, maybe there could be a value-exchange in any form, which does not involve coding on my end :) I would be up for writing docs, I saw that some help could be needed there. 

In regards
Yannis

szabeszg, 2024-04-23:
> I would be up for writing docs

I rarely use this add-on, but when I do, I always struggle to figure out how it works. Especially since, by the time I need to use it again something is always different from its previous state.

Having up-to-date documentation would make a big difference.

kevancress, 2024-04-23:
@Huronsen 

These are great suggestions! I've been meaning to work on a slope dimension, should be fairly straight forward, so I'll try and tackle this when I have some development time!

For the elevation notes, you can actually achieve this currently using [Autofill](https://kevancress.github.io/MeasureIt_ARCH/prop_descriptions/textfields/) on Measureit_ARCH annotations. Here's a quick gif of how to set that up!

[image]

If you are up for writing documentation that would be a huge help, there's definitely some big gaps in the docs that I just haven't had time to cover. Happy to edit and get things published to the documentation pages if your up for writing. On the existing Docs  pages you can hit the "edit on github" to make edits to the raw text for pages if you find anything out of date, or feel free to post drafts of a new page as a github issue.