kevancress/MeasureIt_ARCH #281: custom scales

sampotter123, 2024-06-20:
is it possible to add  a custom scale per dimension?

kevancress, 2024-06-24:
@sampotter123 

Sorry for the slow reply.

Can you explain the use case for this a bit more, I'm not sure I'm grasping it?

My initial thought is that having a unique scale factor per dimension would make it very difficult to tell if any dimension is accurate or not, if you have multiple scales in a single drawing how do you tell which dimensions are at which scale? Seems like this would introduce a lot of uncertainty and opportunity for errors in a drawing... but I may be missing something here?

If you really need to override text per dimension you can enable the "Use Custom Dimension Text" option in the dimensions text fields

kevancress, 2024-08-20:
Closing for now. Reopen with the information requested above if this is still an issue.

closed 2024-08-20 by kevancress