kevancress/MeasureIt_ARCH #282: Awesome plugin - what is the best Blender version ?

99bits, 2024-07-02:
Thank you for this amazing plugin.  Super helpful.

I'm on a Mac, can you suggest what would be the best version of Blender to use.  I'm asking this because it seems that there are many improvements and I'm not sure what is the most updated version of Blender you are using to develop and test the plugin.

Cheers.

kevancress, 2024-07-02:
@99bits I've been using Blender 4.0.1 for my testing. But I think it should work fine up to 4.2

Not sure if the addon will enable with the new "Extensions" system in 4.3... I'll have to look into this.

If you do happen to test with 4.3 let me know how it goes!

Myrkskog, 2024-08-02:
@kevancress Just updated to Blender release 4.2 a51f293548ad
When toggling off viewport overlays, the lines generated by measureit-arch always render in front of objects. This happens with solid viewport shading and cycles. In evee the lines disappear all together. Turning overlays on appears to fix the issue. I tried sorting through the overlays I have set to see if there was a culprit and no luck. Just a full-on toggle of viewport overlays causes this issue.

I am using measureit-arch commit a8b32f4

Please let me know if I can provide any information to you.

[image]

       \\    \\  //      WE|  @ hellfire
        \\    \\//       RD| OS:        nixos 24.05 (uakari)
    ::::://====\\  //    GN| KERNEL:    6.8.9
       ///      \\//     YW| ARCH:      x86_64
  """"//\\      ///""""  BE| UPTIME:    2 hours 32 minutes
     //  \\====//:::::   MA| PACKAGES:  4686
        //\\    \\       CN| SHELL:     zsh
       //  \\    \\      BK| DESKTOP:   hyprland

kevancress, 2024-08-02:
@Myrkskog Yep I'm able to reproduce this on my end too...hmm I wonder if there's been some changes to how the depth buffer gets exposed with EEVEE Next that is causing my depth testing to break... going to make a new issue for this and do some more digging