kevancress/MeasureIt_ARCH #283: Depth Testing Issues in Blender 4.2.0 LTS and Later

kevancress, 2024-08-02:
@kevancress Just updated to Blender release 4.2 a51f293548ad
When toggling off viewport overlays, the lines generated by measureit-arch always render in front of objects. This happens with solid viewport shading and cycles. In evee the lines disappear all together. Turning overlays on appears to fix the issue. I tried sorting through the overlays I have set to see if there was a culprit and no luck. Just a full-on toggle of viewport overlays causes this issue.

I am using measureit-arch commit a8b32f4

Please let me know if I can provide any information to you.

[image]

OS:        nixos 24.05 (uakari)
KERNEL:    6.8.9
ARCH:      x86_64
UPTIME:    2 hours 32 minutes
SHELL:     zsh
DESKTOP:   hyprland (wayland)

_Originally posted by @Myrkskog in https://github.com/kevancress/MeasureIt_ARCH/issues/282#issuecomment-2265845735_

kevancress, 2024-08-02:
Likely related to some change in how the depth buffer is handled with EEVEE Next.

kevancress, 2024-08-20:
Did some basic testing, This same issue is happening with the GPU module drawing examples in the API docs as well: https://docs.blender.org/api/current/gpu.html#mesh-with-random-vertex-colors shows the same issue.

This does appear to be fixed in the latest daily build of Blender 4.3 though @Myrkskog, (In solid view at least, in rendered view MeasureIt_ARCH elements disappear with the overlays when overlays are disabled.).

Fingers crossed this all gets sorted out in the 4.3 release

Myrkskog, 2024-08-21:
Wonderful! Thank you for the heads up. For now I just leave overlays on and toggle what I do not need to see.

Fingers crossed along with you. It will be a nice convenience to avoid a workaround.

Love your plugin! It is a great deal of help and keeps me in Blender. I appreciate not having to export and jump to another software for this feature set.

Myrkskog, 2024-08-22:
[image]
Any idea why the dimensions for axis specific selections draws them offset from the vertices? It appears to use the origin of the world instead of the objects median.

Looking for a way to bypass this. Otherwise I will just photo edit the results for what I need.

kevancress, 2024-08-22:
Ahh, thanks for catching that, had some issues with how I was applying the transform Matrices. Just pushed a fix and made a separate issue for this: https://github.com/kevancress/MeasureIt_ARCH/issues/284

Let me know if that fixes things for you