kevancress/MeasureIt_ARCH #284: Axis Aligned Dimensions not using Object Matrix

kevancress, 2024-08-22:
[image]
Any idea why the dimensions for axis specific selections draws them offset from the vertices? It appears to use the origin of the world instead of the objects median.

Looking for a way to bypass this. Otherwise I will just photo edit the results for what I need.

_Originally posted by @Myrkskog in https://github.com/kevancress/MeasureIt_ARCH/issues/283#issuecomment-2304894491_

kevancress, 2024-08-22:
Fixed in 44ff6103564ab9534b91d258daa261169c36bb5e I think.

@Myrkskog can you give the latest version a test and let me know if this fixes the issue for you?

Thanks!

Myrkskog, 2024-08-23:
Yes it works now! Thank you for the quick fix.
I am using Blender 4.2.1 Hash 396f546c9d82

closed 2024-08-23 by kevancress