kevancress/MeasureIt_ARCH #285: Inconsistence between dimensions in properties and viewports(4.2.1 + 0.5.0)

skonato, 2024-09-03:
Hi Kevan,
I have no idea how to explain the bug and that is why I prepared this video and scene.

https://github.com/user-attachments/assets/7dd7884a-debe-44f1-aba0-281704715281

[BugExample.zip](https://github.com/user-attachments/files/16854605/BugExample.zip)

baldyman01, 2024-09-30:
I'm not with you, inconsistent with what? You have some dimensions and then you're pointing at the settings that defines how far from the mesh the dimensions line is. Two different things.

kevancress, 2024-10-04:
Hey @skonato,

I see what you're getting at, the 1190mm dimension in the bottom left should show up in the list as dimension 4, but instead the list shows dimension 1 twice, with no option to select dimension 4...

This happens because I have to wrap the dimension objects in a generic class so I can display all the different types in the same list, in some cases the wrappers can get out of sync with the actual dimension data.

I should add a simple refresh list operator to solve this. The list wrappers should refresh when dimensions are added or deleted, which is why dimension 4 reappears at the end of your clip after dimension 1 is deleted...

skonato, 2024-10-04:
Hi Kevan,

Yes, you are right. Thanks!)