kevancress/MeasureIt_ARCH #286: Schedule - Should ignore object suffix

baldyman01, 2024-09-18:
The schedule generator has an option of group rows and total them up which is great, except that it will never have an identical row for something line RNA object name unless it can be set to ignore the suffix -

Name,Count
40 mm  Waste Pipe.003,1
15mm Wall Mount Bend.005,1
15mm Wall Mount Bend.006,1
Cylinder.016,1

Should be -

Name,Count
40 mm  Waste Pipe,1
15mm Wall Mount Bend,2
Cylinder,1

Cheers
Pete

Blender 4.2, Measureit_ARCH version, from about  a week ago.

kevancress, 2024-09-18:
@baldyman01 My thought on this was that in cases where your collecting / counting a number of instances of the same object then those objects should have linked mesh data. In that case changing the RNA prop field to `.data.name` instead of just `.name` should give the result your looking for. 

Here's a demo setup:

[image]

Which results in

Name,Count
Truncated Octahedron,6
Cube,4

kevancress, 2024-09-18:
I could look at adding an option to ignore the .### suffix too, but I think that risks grouping objects might have accidentally ended up with the same object name, but aren't actually the same thing (i.e. don't share the same data)

baldyman01, 2024-09-18:
Maybe I didn't think this through completely. So I have two use cases, one is instances of identical objects, lets say a 15mm Pipe Elbow. So using .data.name sorts that out. 

I then also have multiple pieces of pipe, of differing lengths. So this ends up as Pipe, Pipe.001 etc. These are not the same data, but are all the same type of object. So 3 elbows connecting 5 pieces of pipe ends up with 5 separate pipe entries.  

Being able to ignore the prefix would allow a single item with a count in this instance. But I'm now wondering how useful that actually is. What you really want is one item with a length sum, but that gets a lot more complex.

Maybe it's easiest to keep it as is, add a length column and total it up later in the spreadsheet.

kevancress, 2024-09-19:
Ahh okay... yeah the schedules are a bit limited for tasks like that right now... I should give some thought to adding more control over defining categories, and which properties get summed, how to add subtotals for properties etc...

Ideally I could imagine for your pipe use case being able to generated something like this might be ideal?

| Category   |      Name      | Count | Length|
|----------|:-------------:|------:|:----:|
| **Elbows** |   |  | 
|  |  90 degree elbow | 3 | N/A
|  |  45 degree elbow | 3 | N/A
|**Pipes** |
|  | 0.5m Pipe  | 2 | 0.5m|
|  | 0.25m Pipe | 3 | 0.25m|
|  |  | **Length Total**| 1.75m|

I'll try and give this some thought over the next couple days.

VeaseyB, 2024-11-01:
Apologies, this is unrelated, but I am an MEP engineer who aspires to use Blender for piping models and drawings.  Are you using Geo Nodes to size your pipe fittings?  How do you find snapping to pipe ports?  Any advice would be gratefully received.  Thanks.

baldyman01, 2024-11-01:
No. I have an asset library of pipe sizes and fittings that I drag and drop. Aligning is tricky as Blender doesn't have a connector system like, say, Visio. 

If you place the origins of the pipes and fittings in the right places then most of the work can be done with the align tool. I keep align X,Y and Z mapped to top numbers keys 1,2 and 3. You can align in two axes by hotkey then snap to place in the third axis. This makes pipes and 90 degree joints easy, different angle elbows have to be aligned by eyeball though, so not perfect but you get used to it quite quick.