kevancress/MeasureIt_ARCH #288: Annotation Linked To Custom Property Value

VeaseyB, 2024-11-01:
I am trying to create an annotation with its text linked to the value of a Custom Property (Property Name=Prop, Value (as a string)=Custard).  I change the Annotation to report a Custom Property and put [<Prop].  But Blender displays 'Bad Data Path'.  What should the syntax be?  Thanks

[Test.zip](https://github.com/user-attachments/files/17600276/Test.zip)
[image]

kevancress, 2024-11-01:
@VeaseyB Thanks for reporting this.

2 things here:

1) the syntax for this should be `["prop"]` . I'll update the docs to make the syntax more clear (the < in the docs is a typo)

2) There was actually a bug in how custom properties were getting evaluated, so the correct syntax wouldn't have been working either. I've fixed that now in f988c6b9b2aef997da46bdb3d84ce7db9830d7d5

If you download the latest version now and use the correct syntax things should be working properly now!

VeaseyB, 2024-11-01:
That works great now, thank you!

kevancress, 2024-11-01:
Great! Glad It's working!

closed 2024-11-01 by kevancress