kevancress/MeasureIt_ARCH #290: area and angle don't run with reprojected coordinates with blender gis

bonushenricus, 2025-01-05:
Hello and happy new year.
It seems that area and angle have some problem when I'm trying with blender after the use of blender GIS.
I tried with simple cube added to two 3d view, one without use of blender gis and one with the previous use of blender gis.
The annotations and dimension lines was not affected.
Best wishes

bonushenricus, 2025-01-06:
I use 0.5.0 dev 7967b29 on Blender LTS 4.2.4
The problem is not Blender-GIS, the area and angle was positioned in other place than the view also when the coordinate are not projected

AdrianoVM, 2025-02-19:
Same here, using Blender 4.3.2.
Area, Angle, and Arc do not follow object origin, and instead stay relative to 0.
This is quite annoying as it forces you to apply location with `Ctlr+A -> Location`. Which makes moving and rotating objects more difficult.

Here is an example:
[image]

kevancress, 2025-03-04:
Hey @bonushenricus @AdrianoVM 

Sorry for the slow fix, been a busy start to the year! This should be resolved in the latest commit 21bb7025c6cd9bbf2221fe1f9e91d25d3d2065ca

Let me know if this fixes things for you. Thanks for reporting this issue!

AdrianoVM, 2025-03-10:
It indeed seems to be fixed. Area, Angle, and Arc appear at the correct spot, and the object can be moved, scaled and rotated without issues. 
Thank you for the fix !

closed 2025-04-25 by kevancress